CircuitVerse/cv-frontend-vue #501: 🐞 Bug: The navbar dropdown's z-index is lower than the tab bar, causing the first option of the dropdown menu to be hidden.

nickhil-verma, 2025-03-03:
### Is there an existing issue for this?

- [x] I have searched the existing issues.

### Describe the bug

##### The navbar dropdown's z-index is lower than the tab bar, causing the first option of the dropdown menu to be hidden.

### Steps to Reproduce

1. Go to simulator 
2. hover/click over dropdowns 

### Expected Behavior

The dropdown menu is expected to have a higher z-index to ensure its options are not hidden.

### Screenshots

https://github.com/user-attachments/assets/6f9bae26-c4e7-4ec5-83e6-452ffbbd57d7

### Is the faced issue/bug related to the Vue simulator?

Yes

### Used Vue simulator with or without backend?

None

### Is the bug present only on the dev server, the build, or both?

None

### Device Information

```markdown

```

### Are you working on this issue?

Yes

niladrix719, 2025-03-04:
already fixed in latest

closed 2025-03-04 by niladrix719

nickhil-verma, 2025-03-04:
> already fixed in latest
oops i didn't saw that.