CircuitVerse/cv-frontend-vue #506: Bug: Circuit Elements Drag Outside Workspace and Have Incorrect Z-Index

gitsofaryan, 2025-03-06:
### Description

When dragging elements such as Testbench and Timing Diagram, they are able to move outside the visible workspace (canvas area). Additionally, these elements have a lower z-index than the top navbar and the right-side properties panel, which causes them to get partially or fully hidden behind these UI components.

### Steps to reproduce

- Open the simulator.
- Drag the Testbench or Timing Diagram element towards the top or right edges.
- Observe that the element can move out of the main canvas area.
- Also notice that these elements go behind the navbar and properties panel instead of staying above them.

### Expected Behavior

- Elements should stay confined within the canvas/workspace area.
- Elements should have a higher z-index than the navbar and properties panel to remain visible during positioning.

### Actual Behaviour

https://github.com/user-attachments/assets/3f420cc0-d316-4620-871a-350b4619f273

gitsofaryan, 2025-03-06:
cc: @vedant-jain03 @niladrix719 @aryanndwi123

vedant-jain03, 2025-03-07:
@gitsofaryan thanks, go for it!

harikrishna-au, 2025-03-11:
@gitsofaryan  is you done for the PR?else Can i help with this issue