CircuitVerse/cv-frontend-vue #509: Docs: Add Documentation for Desktop application Development

niladrix719, 2025-03-07:
add prerequisites and commands to run dev/build in the readme

niladrix719, 2025-03-07:
lets keep this for someone else

gitsofaryan, 2025-03-07:
hi @niladrix719 , i have completed the spec yml issue, can i pls work on this.

ThatDeparted2061, 2025-03-07:
@gitsofaryan Let some newcomer do this one, They'll get to understand the code base too

ThatDeparted2061, 2025-03-07:
@gitsofaryan you can(/should) collaborate with someone new and together you guys can do this one.

Kartikayy007, 2025-03-10:
I think I can work on this @niladrix719  @ThatDeparted2061

ThatDeparted2061, 2025-03-10:
Yeah go ahead @Kartikayy007, take your time and take the help from @gitsofaryan.

gitsofaryan, 2025-03-10:
@Kartikayy007 we can collaborate to build most optimal doc of desktop tauri simulator. Let me know if you are open for collaborations!!

Kartikayy007, 2025-03-12:
@niladrix719  @gitsofaryan you provide more info about the issue? I'll raise a PR again

gitsofaryan, 2025-03-12:
@Kartikayy007 lets collab?

gitsofaryan, 2025-03-12:
@Kartikayy007 refer this https://github.com/gitsofaryan/cv-frontend-vue/tree/Doc?tab=readme-ov-file
i have added steps would be happy if you collaborate and enhance accordingly.

niladrix719, 2025-03-12:
commands are `npm run tauri dev` and `npm run tauri build`

Kartikayy007, 2025-03-13:
@niladrix719 @ThatDeparted2061 please review

Kartikayy007, 2025-03-15:
@niladrix719 please review

blazethunderstorm, 2025-06-14:
@Arnabdaz pls review the pr

ViRUS-0-0, 2025-12-18:
@niladrix719 Can you please assign this issue to me?

NishuRajput12, 2026-02-24:
Hi @samarth-swami, I’d like to help enhance this documentation. Can we collaborate or split tasks?

Amrit-raj50, 2026-09-06:
Hi Team, I see this issue was reopened because the previous PR was merged into a fork rather than the main repo. I'd like to work on this and submit a proper PR that adds the desktop development documentation directly to this repository. Could you please assign this to me?