CircuitVerse/cv-frontend-vue #525: Tauri Desktop Application Enhancement: Update Web-Based Dialogs to Desktop-Based Dialogs

gitsofaryan, 2025-03-12:
### Descrption

The CircuitVerse desktop app, built with Tauri framework, currently uses web-based dialogs that display messages like "We have saved your project: aj in your browser's localStorage" when saving a project, as seen in the screenshot. This is misleading in a desktop context because the app should save files to the local storage rather than the browser’s localStorage. when using desktop we need to update the save dialog to display "We have saved your project: [projectName] in local storage," where "local storage" refers to the desktop filesystem.

[image]

ThatDeparted2061, 2025-03-13:
Yeah that'll be better you should fix it.

blazethunderstorm, 2025-06-07:
@ThatDeparted2061 is this resolved can i work on this ?