CircuitVerse/cv-frontend-vue #529: Feature  Guide for new comer to desktop

malviya-rajveer, 2025-03-13:
### Is your feature request related to a problem? Please describe

in our web application when we start simulator very first time a nice guide is given for what are the things on the simulator window. If we can add this feature to our desktop app it would be nice and great help for new comers.

[image]
[image]

### Describe the solution you'd like

the user gets guide of what are the things are there on the window he is going to use

### Describe alternatives you've considered

no

### Is requested feature related to vue simulator?

yes

### Are you working on this?

Yes

niladrix719, 2025-03-14:
This feature is already there for desktop app

malviya-rajveer, 2025-03-14:
Oops, I searched for it later. My bad...

closed 2025-03-14 by malviya-rajveer