CircuitVerse/cv-frontend-vue #535: 🐞 Bug: Typescript error in Navbarlink2 file

malviya-rajveer, 2025-03-14:
### Is there an existing issue for this?

- [x] I have searched the existing issues.

### Describe the bug

The logixFunction in Navbarlink2.vue has type any implicitly which is not safe and not good practice and giving red squiggles and can create issues further 

### Steps to Reproduce

1. go to data.js file
2. Make it javascript to typescript file
3. give type to the logicFunction

### Expected Behavior

the logicFuntion would have specific type and not any 

### Screenshots

[image]

### Is the faced issue/bug related to the Vue simulator?

No

### Used Vue simulator with or without backend?

None

### Is the bug present only on the dev server, the build, or both?

Both

### Device Information

```markdown
os:Ubuntu20.04.4
Browser: Chrome
```

### Are you working on this issue?

Yes

malviya-rajveer, 2025-03-14:
I have solved this issue. should I create the PR @niladrix719

niladrix719, 2025-03-14:
yeah

malviya-rajveer, 2025-03-15:
please preview it @niladrix719

niladrix719, 2025-03-15:
https://github.com/CircuitVerse/cv-frontend-vue/issues/414

closed 2025-03-15 by niladrix719