CircuitVerse/cv-frontend-vue #539: 🐞 Bug: Typescript Types of parameters missing in the function and one parameter declared but not read

malviya-rajveer, 2025-03-15:
### Is there an existing issue for this?

- [x] I have searched the existing issues.

### Describe the bug

In this image, there are no types given to the parameters and the prompt was also throwing error
[image]
here I solved the other things and then I saw the scope parameter is not being read anywhere. Therefore, I would like to ask @niladrix719 Should we remove it?

[image]

### Steps to Reproduce

1. Go to src/components/DialogBox/CombinationalAnalysis.vue
2. find createBooleanPrompt function 
3. solve the issue

### Expected Behavior

the code should be strictly type checked and error free 

### Is the faced issue/bug related to the Vue simulator?

Yes

### Used Vue simulator with or without backend?

None

### Is the bug present only on the dev server, the build, or both?

None

### Device Information

```markdown

```

### Are you working on this issue?

Yes

niladrix719, 2025-03-15:
you can put you pr referencing this issue https://github.com/CircuitVerse/cv-frontend-vue/issues/414

closed 2025-03-15 by niladrix719

malviya-rajveer, 2025-03-15:
okay thankyou!!