CircuitVerse/cv-frontend-vue #557: Feature :  Authentication for simulator - Web and Tour

ThatDeparted2061, 2025-05-11:
### Is your feature request related to a problem? Please describe

## Implement Authentication for web and Tauri simulator

Status - added authentication with a separate pop up side desk, handled apis for login and register, improved the authstore handling

To do - cudnt recheck the apis for production (check the comments in the PR), handle dashboard (need brainstorming here) 

### Describe the solution you'd like

Raised A PR.

### Is requested feature related to vue simulator?

Yes

### Are you working on this?

Yes

ThatDeparted2061, 2025-05-11:
@vedant-jain03 @aryanndwi123 @niladrix719

blazethunderstorm, 2025-06-04:
@niladrix719 is this issue open to work on ?

blazethunderstorm, 2025-06-04:
@salmoneatenbybear sorry for not looking up to that would take care next time

closed 2026-01-06 by tachyons