CircuitVerse/cv-frontend-vue #587: Feature: Add Clock Divider Component

blazethunderstorm, 2025-06-10:
### Is your feature request related to a problem? Please describe

CircuitVerse currently provides a basic Clock component with a fixed frequency. However, users often need slower clock signals to visually follow the behavior of sequential circuits like counters, FSMs, and flip-flop chains. Manually building a clock divider each time using multiple flip-flops is repetitive and not beginner-friendly.

### Describe the solution you'd like

I would like to contribute a built-in Clock Divider component to CircuitVerse. This component would take a Clock input and output a slower Clock signal based on a user-defined divide factor (e.g., divide-by-2, 4, 8). Internally, it can use a flip-flop-based counter or toggle logic. This will help simplify timing visualization for educational projects and reduce the need to build repetitive divider circuits manually.

### Is requested feature related to vue simulator?

Yes

### Are you working on this?

Yes

KumarADITHYA123, 2025-11-29:
"Hi! I have completed the work for this issue. I've implemented the Clock Divider component, which allows users to easily configure the division factor (e.g., ÷2, ÷4, ÷8).

I have raised a Pull Request with the changes. Could you please review it?

ShinichiShi, 2026-02-22:
@092vk could you please validate this feature request