CircuitVerse/cv-frontend-vue #610: 🐞 Bug: Wires behaving abnormally — misalignment, unwanted node creation, and glitchy convergence

blazethunderstorm, 2025-06-18:
### Is there an existing issue for this?

- [x] I have searched the existing issues.

### Describe the bug

When you connect wires between two points, they sometimes move or break even if you didn’t do anything wrong. They also create extra dots (nodes) for no reason. This makes the circuit look messy and confusing.

### Steps to Reproduce

1)Open the circuit simulator.

2)Add two points (nodes) to the screen like use gate and then connect inputs and outpu

3)Connect them with a wire.

4)Try clicking or dragging near the wire.

You’ll see the wire breaks or adds extra dots (nodes) by itself.

### Expected Behavior

https://github.com/user-attachments/assets/35b8734c-487c-4fbe-ba0f-94f2e5e16f6f

### Screenshots

https://github.com/user-attachments/assets/4a1d4d70-42c1-44e0-96d7-37409a861fa0

### Is the faced issue/bug related to the Vue simulator?

Yes

### Used Vue simulator with or without backend?

None

### Is the bug present only on the dev server, the build, or both?

None

### Device Information

```markdown
OS: Ubuntu 
Browser: Chrome
```

### Are you working on this issue?

Yes

blazethunderstorm, 2025-06-18:
@ThatDeparted2061 @Arnabdaz will makes its pr in a time

ThatDeparted2061, 2025-06-18:
Okay